hyc-sg/Helios-public #8: Inverter log: file not found

HYC-Alice, 2022-05-17:
**Machine ID**
SA031

**Describe the bug**
The inverter log data for May 12th-May 16th are missing. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to FB_InverterDataLog and put enable to false.

**Expected behavior**
We expect that the inverter data log would record the file and we would be able to access it regardless of the time we open up the data.

**Images or Screenshots**
Can't find the inverter data from May 12th-today. Encountered error with 1804
[image]

**Additional context**
Was able to reset it by manually enabling it. Have data for May 16th 9:44am onwards.

enzo-hycsg, 2022-05-17:
Bug acknowledge. I'm working on this...

enzo-hycsg, 2022-06-07:
Beckhoff can't replicate this bug. To address this, a containment was implemented in version 2.12, if an error is detected,  the function block will disabled and then auto enabled.

closed 2022-07-20 by HYC-Alice