hyc-sg/Helios-public #12: Heating Fault bug

simchu, 2022-07-06:
**Machine ID**
SA034

**Describe the bug**
Zone 6 temperature higher then set temperature PLC control the power not on and got heating fault

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Main Page'
2. Click on 'Select recipe Yuma Power Bowl. Turn on heater and run overnight'
3. Run at 0 ft/min
4. Bug present after zone 6 heated up more than SV125 which is 136 continuously. the fault appear.

**Expected behavior**
A clear and concise description of what you expected to happen.
Zone 6 heated up more than SV125 which is 136 continuously. the fault appear.

**Images or Screenshots**
Please try to capture images or screenshots of the bug for better understanding and resolution.
[image]

**Additional context**
Add any other context about the problem here.
MCB doesn't trip

HYC-Alice, 2022-08-17:
With Issue #13: Beckhoff has no solution to this issue yet. Requesting for their presence on-site.

HycAlex, 2022-09-06:
Zone 6 set point is 125 which is 40 degree below zone 5(165) and same value at zone 7(125). at statics condition (conveyor at rest) heat distribute from zone 7 gradually rise to zone 5 and thus not possible to control zone 6 temperature same set point as zone 7 and 40 degree below zone 6. 
Nothing to do with beckhff, this is physics

closed 2022-09-06 by HycAlex