hyc-sg/Helios-public #19: CIP: Motion Speed Offset Range

HYC-Alice, 2022-09-19:
**Machine ID**
SA031 and for rest of alpha systems

**Describe the bug**
On 8 Sep, we changed the offsets for RS due to observed high current, loud sounds and waves. FP Gavin has requested us to conduct a motion DOE and come up with a set range of acceptable speed offsets adjustments. 

**To Reproduce**

**Expected behavior**
1. A range of acceptable speed offsets adjustments WI 
2. When to change offsets/jackshafts; What would be the expected outcome if a certain offset was adjusted.. What to look out for, what motor to adjust.

**Images or Screenshots**

**Additional context**
Per Gavin's request: All offset changes need to request permissions from FP. And only level 3 trained/HYC Control Personnel should be able to change the offsets.

enzo-hycsg, 2022-10-30:
Any update on this?

HYC-Alice, 2022-11-09:
> Any update on this?

Compiled the alpha systems current offsets for all speed.
As of now, we only recently changed SA031's 60fpm and 105fpm RS offsets.
Then yesterday (8Nov), we changed SA035's LS 30fpm offsets. 
Still monitoring, but am working on WI.

closed 2023-05-10 by enzo-hycsg