hyc-sg/Helios-public #22: 6Sep: LS Drum Encoder Fault

HYC-Alice, 2022-09-20:
**Machine ID**
SA035

**Describe the bug**
Encountered LS Drum Encoder Fault message on SA035. Found LS infeed coupling opened, and drum encoder disc shifted down.

**To Reproduce**

**Expected behavior**

**Images or Screenshots**
[image]
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

enzo-hycsg, 2022-10-11:
This is a valid bug. Bug fix to be release on v3.2, WW42.5

enzo-hycsg, 2022-10-14:
For software release v3.2, WW42.5

enzo-hycsg, 2022-10-30:
Release on v3.2

closed 2022-10-30 by enzo-hycsg