hyc-sg/Helios-public #28: SA036's event log and alarm log are mismatch again for log9

HYC-Alice, 2022-10-11:
**Machine ID**
SA036

**Describe the bug**
SA036's event log and alarm log are mismatch again for log9.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Oven-->Log
2. Click on Alarm Log/Event Log folder and look at Log 9
[image]

**Expected behavior**
Expected Log 9's event log/alarm log to not have any log as the machine wasn't running/touched on 9Oct.

**Images or Screenshots**
Log9 has log 10’s log(alarm)
[image]

Then log10 is like this (alarm)
[image]

Event is the same issue.. log 10 events for log 9..
[image]

Log10's event log didn’t have headers
[image]

**Additional context**
There wasn’t any shutdown. The machine was just not running at all yesterday (9Oct), then the log happened when we ran the machine today (10Oct)

enzo-hycsg, 2022-10-11:
This is a valid bug. Issue will be escalated to Beckhoff, the issue is related to "Logged day" logger inputs. The function block manage it's own logged day based on system date and time.

enzo-hycsg, 2022-10-28:
Closed and keeping in view this bug as BECKHOFF did not find any errors on our code. We'll reopen once we have critical occurrences of this bug.

closed 2022-10-28 by enzo-hycsg