hyc-sg/Helios-public #31: SA036-Heater Health (Ampere Reading) is a bit faulty on HMI

HYC-Alice, 2022-11-09:
**Machine ID**
SA036

**Describe the bug**
SA036 has two faulty heaters with their fuse being lifted up.
H21, H41(heater fuse burnt), but on the hmi it’s not showing that it’s faulty. 
For H21, the HMI ampere reading showed ~26A (which is in the +/-2A range), thus I can see why it's not "red" on the HMI. 
For H41, we have a heater fuse burnt, and the HMI ampere reading showed ~24A and you can see the number of "heaters" counter on the HMI is at 3 heaters, but still doesn't light up red. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Conveyor and Heater Runtime, force on all heaters
2. Check control box ampere reading as well

**Expected behavior**
Expected H21, H41 to have a red fill. 

**Images or Screenshots**
[image]

[image]

**Additional context**
Add any other context about the problem here.

enzo-hycsg, 2022-11-16:
Try swapping CT or EL3446 cards, it's likely a hardware issue.

HYC-Alice, 2022-12-18:
Update: MX4 site FSE personnel conducted some tests with CT, came to the conclusion that will need to change EL3446 cards.

closed 2023-01-13 by enzo-hycsg