hyc-sg/Helios-public #32: 31Oct: SA036 conveyor runtime and heater runtime restarted after replacing Z5 SSR

HYC-Alice, 2022-11-09:
**Machine ID**
SA036

**Describe the bug**
 SA036 conveyor runtime and heater runtime restarted after replacing Z5 SSR. It didn't record the "last cleared" as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Replace SSR 
2. Restart machine

**Expected behavior**
SA036 conveyor runtime and heater runtime shouldn't be cleared/reset. Should still have data since Sep 9th.

**Images or Screenshots**
Please try to capture images or screenshots of the bug for better understanding and resolution.

31Oct: 8:38 am before SSR Z5 replacement 
[image]

31Oct: 10:30am after SSR Z5 replacement
[image]

**Additional context**
Add any other context about the problem here.

HYC-Alice, 2022-11-09:
Updated on 8Nov: Conveyor and heater runtime are added.
[image]

enzo-hycsg, 2022-11-16:
This is WritePersistent data function bug where we can't control... data corruption is common to PC like controllers during a sudden shutdown. Counter protection are already in place, runtime data is being log hourly for traceability such as this, a special data backup function is also in place wherein data is being written into memory within 1 second during a sudden shutdown, this function only works if the IPC battery is ok. Ultimately, always practice to properly shutdown the IPC.

closed 2022-11-16 by enzo-hycsg