hyc-sg/Helios-public #33: CIP: Adding another password for "Expert Use"

HYC-Alice, 2022-11-22:
**Machine ID**
SA031/ All alpha or future software 

**Describe the bug**
FP Gavin say he wants to keep expert use only locked under another password as he doesn't want the offsets to be easily touched by others. 

**To Reproduce**
Steps to reproduce the behavior:
Login as Engineer

**Expected behavior**
A clear and concise description of what you expected to happen.

**Images or Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

enzo-hycsg, 2022-12-09:
This feature will be accessible through "Admin" user only.

dc3nity, 2022-12-19:
@enzo-hycsg to confirm, this will be visible for all users, but grey out and no interaction for any users except admin users?

enzo-hycsg, 2023-01-31:
Yes correct.. :)

dc3nity, 2023-01-31:
Updated and plan release for ER5

closed 2023-01-31 by dc3nity