hyc-sg/Helios-public #35: Wrong tooltip info for 105FPM start button

dc3nity, 2022-12-19:
**Machine ID**
ALL ALPHA SERIES

**Describe the bug**
105fpm start button tooltip displays as 120fpm start instead of 105fpm

**To Reproduce**
Steps to reproduce the behavior:
1. Reset and ready the machine
2. Mouse over (do not click) 120fpm start
3. "Conveyor Start @ 120fpm" displayed

**Expected behavior**
"Conveyor Start @ 105fpm"

**Images or Screenshots**
Please try to capture images or screenshots of the bug for better understanding and resolution.

**Additional context**
Add any other context about the problem here.

dc3nity, 2022-12-19:
@enzo-hycsg request to self-assign and resolve in 3.4

enzo-hycsg, 2023-01-13:
@dc3nity how is this? I prefer to remove the tooltip text for object that are self-explanatory or has a name on it. Let's use tooltip text where object has abbreviated text or difficult to understand.