hyc-sg/Helios-public #36: 20Dec: SA032 Faulty reading for all phases in Zone 8

HYC-Alice, 2022-12-20:
**Machine ID**
SA032

**Describe the bug**
Discovered a faulty reading for all phases in Zone 8 at SA032. 

**To Reproduce**
Steps to reproduce the behavior:
1. Login to engineer mode
2. Click on '!'
3. Turn heaters on 
4. Bug present

**Expected behavior**
The heater counter should only have "4" 

**Images or Screenshots**
[image]
[image]
[image]

**Additional context**
Add any other context about the problem here.

enzo-hycsg, 2023-01-13:
@HYC-Alice this is likely a hardware issue, how the reading right now?

enzo-hycsg, 2023-02-15:
@HYC-Alice just checking in on the status of this, has this been rolled out as hardware issue?

closed 2023-03-09 by HYC-Alice