hyc-sg/Helios-public #53: Conveyor Belt Speed Change

HYC-Alice, 2023-07-11:
**Machine ID**
SA031/alpha system

**Describe the bug**
Footprint Mike Miles said, for the Conveyor Belt Speed change, they would like a software revision that permits the user to set the belt speed anywhere between 30-105.

Currently we have 3 speeds, 30, 60, 105. They want to be able to adjust the conveyor belt to run a bit faster than their infeed product transfer belt. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Run at ...
4. Bug present

**Expected behavior**
A clear and concise description of what you expected to happen.

**Images or Screenshots**
Please try to capture images or screenshots of the bug for better understanding and resolution.

**Additional context**
Add any other context about the problem here.

HYC-Alice, 2023-07-26:
Please see below for the high level of the project request:

1.	Seeking the ability to run at different belt speeds between the existing 30,60,105 FPM options as R&D tools for process optimization rather than an existing development project need
a.	New additions are proposed at 40, 50, 70, 80, 90 FPM
2.	The assumption is that engineers and techs will be working with the AZ HYCO as an R&D tool, differences in HMI should lead to negligible downtime from input error
a.	To pursue further: conversion of HMI from temp recipe + belt speed to a combination of the two into a single selectable “formula” for HVM use in MX
3.	Current need: Upfield 216 g tubs may be denested and sprayed faster (160  UPM) than the current temp (max 150 C) and belt speed (60 FPM) can accept (145 UPM) without manual laning of parts. Increase to 70 FPM at the same temps would maximize throughput of the upstream equipment. More opportunities may be available with future products
4.	Collaboration in development is still desired in the same capacity as current; these additions would add to the engineering tools at our disposal for the optimization of manufacturing processes

HycAlex, 2023-07-27:
proposal #1
1. include speed 30 to 105FPM (step of 10) offsets in XML file
2. include speed selection in recipe - pull down selection - by Engineer User group login
3. withdraw speed offset upon recipe downloaded 
4. remove speed 30, 60 and 105 Start button. consolidate into ONE start button for each silo
5. operation seq: download recipe, Heater on, Temperature ok, conveyor start
Comments?